xanderhendriks/docker-stm32cubeide #3: Example project request

justinweinberg, 2023-11-07:
Hi,

I love the idea of being able to build within a container and appreciate the time and effort that has gone into this docker image.  Unfortunately, I am unable to get the command line build working on our STM32 project.

Does anyone have a simple example STM32IDE project that works with a version of Xander's container + command line arguments passed?  Blinky style project would be great.

Thank you.

xanderhendriks, 2023-11-07:
Hi Justin,

Please have a look at my example project which builds it in the cloud using github actions: https://github.com/xanderhendriks/stm32-sample-application if you would like to run the commands manually then you can see them in the action repos [entrypoint.sh](https://github.com/xanderhendriks/action-build-stm32cubeide/blob/master/entrypoint.sh)

Cheers,
Xander

xanderhendriks, 2023-11-07:
I have also added usage details to the [readme](https://github.com/xanderhendriks/docker-stm32cubeide#usage)

justinweinberg, 2023-11-07:
Perfect thank you!!

closed 2023-11-07 by justinweinberg