xanderhendriks/docker-stm32cubeide #5: More detailed instructions on how to get started

krupis, 2024-02-28:
Hello. I am working with the STM32 project and we would like to setup CI/CD pipeline to automatically build and generate .hex artifact upon each release. I have used esp-idf docker image before to do the same thing on a different MCU (ESP32) using the espressif/idf:v5.0.4 image. My .gitlab-ci.yml was as following:

```
esp_idf_generate_build_files:
  image: espressif/idf:v5.0.4
  variables:
    GIT_SUBMODULE_STRATEGY: normal
  stage: build
  rules:
    - if: $CI_COMMIT_TAG # Run this job when a tag is created
  before_script:
    - idf.py build
  script:
    - mkdir binaries;
    - mkdir full_binary;
    - cd build; 
    - echo ${CI_PROJECT_NAME}
    - esptool.py --chip esp32 merge_bin -o merged_fw.bin --flash_mode dio --flash_freq 40m --flash_size keep 0x1000 bootloader/bootloader.bin 0x10000 ${CI_PROJECT_NAME}.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin 0xf000 phy_init_data.bin
    - cd ..
    - cp build/${CI_PROJECT_NAME}.bin binaries/${CI_PROJECT_NAME}-${CI_COMMIT_TAG}.bin
    - cp build/merged_fw.bin binaries/${CI_PROJECT_NAME}-${CI_COMMIT_TAG}_full.bin

    - cd binaries
    - zip ${CI_PROJECT_NAME}-${CI_COMMIT_TAG}_binaries.zip *.bin # zip all .bin files
    - ls -la
    - cd ..
    - 
    ```

The whole STM32 project is STM32CubeIDE based. After a quick search, I have found your docker image but with my limited docker knowledge I cannot fully figure out how to get this whole thing running as there seems to me that some steps are missing.

On my Ubuntu virtual machine, I have installed your docker image using the following command:

```
my_user@xxx-gitlab:~$ docker pull xanderhendriks/stm32cubeide:7.0
permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post "http://%2Fvar%2Frun%2Fdocker.sock/v1.24/images/create?fromImage=xanderhendriks%2Fstm32cubeide&tag=7.0": dial unix /var/run/docker.sock: connect: permission denied
my_user@xxx-gitlab:~$ sudo docker pull xanderhendriks/stm32cubeide:7.0
7.0: Pulling from xanderhendriks/stm32cubeide
3b65ec22a9e9: Pull complete
10bf1cfff187: Pull complete
c13f6adea5ad: Pull complete
75133ffded2c: Pull complete
2832732aa9fd: Pull complete
Digest: sha256:3724d7ae4ea820ea75e6e2037bbd4d257b87190496f0289bbd86a0b6ecaaec21
Status: Downloaded newer image for xanderhendriks/stm32cubeide:7.0
docker.io/xanderhendriks/stm32cubeide:7.0
```

Additionally, I have cloned my STM32 project from the Gitlab to my Ubuntu virtual machine ( /tmp/test_project directory ):
```
my_user@xxx-gitlab:/tmp/test_project$ ls
my_stm32_project
```

I cannot fully figure out what are the next steps in order to build this project:
How to correctly run the docker image and what arguments do I need to pass? 
How to build the project?
How to setup everything within the CI/CD script?

I would highly appreciate if you could clarify.

xanderhendriks, 2024-02-28:
I think you'll find this [sample application](https://github.com/xanderhendriks/stm32-sample-application/blob/master/.github/workflows/continuous_integration_pipeline.yml#L69) gives you the information you need.
The [github action](https://github.com/xanderhendriks/action-build-stm32cubeide) uses this docker under the hood. [Here](https://github.com/xanderhendriks/action-build-stm32cubeide/blob/master/entrypoint.sh) you can see how to build from the command line in the docker.

krupis, 2024-02-28:
@xanderhendriks 
Thanks for a quick response. Before I try to integrate this to my Gitlab CI/CD pipeline, I will try to build from the command line as you have suggested.

So I have tried the following:

**1. Run docker container:**
```
my_user@xxx-gitlab:~$ sudo docker run -it --rm=true xanderhendriks/stm32cubeide:7.0
[sudo] password for my_user:
root@66b51abe60ef:/# ls
bin  boot  dev  etc  home  lib  lib32  lib64  libx32  media  mnt  opt  proc  root  run  sbin  srv  sys  tmp  usr  var
root@66b51abe60ef:/#
```

2. Try to execute the command as you have suggested:
```
root@66b51abe60ef:/# stm32cubeide --launcher.suppressErrors -nosplash -application org.eclipse.cdt.managedbuilder.core.headlessbuild -data /tmp/stm-workspace -import "$1"
Project: file:/ can't be found!
Stm32cubeide:
JVM terminated. Exit code=1
/opt/st/stm32cubeide_1.10.1//plugins/com.st.stm32cube.ide.jre.linux64_2.2.0.202203101208/jre/bin/java
-Dosgi.requiredJavaVersion=1.8
-Dosgi.instance.area.default=@user.home/STM32CubeIDE/workspace_1.10.1
-Declipse.buildId=Version 1.10.1
-DproductMaturityGrade=mm
-XX:+UseG1GC
-XX:+UseStringDeduplication
-Xms256m
-Xmx1024m
-jar /opt/st/stm32cubeide_1.10.1//plugins/org.eclipse.equinox.launcher_1.6.400.v20210924-0641.jar
-os linux
-ws gtk
-arch x86_64
-launcher /opt/st/stm32cubeide_1.10.1/stm32cubeide
-name Stm32cubeide
--launcher.library /opt/st/stm32cubeide_1.10.1//plugins/org.eclipse.equinox.launcher.gtk.linux.x86_64_1.2.400.v20211117-0650/eclipse_11600.so
-startup /opt/st/stm32cubeide_1.10.1//plugins/org.eclipse.equinox.launcher_1.6.400.v20210924-0641.jar
--launcher.overrideVmargs
-exitdata 3
-application org.eclipse.cdt.managedbuilder.core.headlessbuild
-data /tmp/stm-workspace
-import
-vm /opt/st/stm32cubeide_1.10.1//plugins/com.st.stm32cube.ide.jre.linux64_2.2.0.202203101208/jre/bin/java
-vmargs
-Dosgi.requiredJavaVersion=1.8
-Dosgi.instance.area.default=@user.home/STM32CubeIDE/workspace_1.10.1
-Declipse.buildId=Version 1.10.1
-DproductMaturityGrade=mm
-XX:+UseG1GC
-XX:+UseStringDeduplication
-Xms256m
-Xmx1024m
-jar /opt/st/stm32cubeide_1.10.1//plugins/org.eclipse.equinox.launcher_1.6.400.v20210924-0641.jar
root@66b51abe60ef:/#
```

I am still failing to understand how docker supposed to get the project? Do I need to clone my project inside the docker image, then cd into the project directory and run the command?

It is important to note that I am actually writing code on my local machine. The Virtual machine that is running the docker container does not know anything about the project. That is my main confusion.. Do you understand what I mean?

krupis, 2024-02-28:
UPDATE.

I have figured out how to build using command line. The key is to run docker and specify -v argument to bind the docker container.

For example:
```
sudo docker run -it --rm=true -v /tmp/test_project/my_project:/tmp/test_project/my_project xanderhendriks/stm32cubeide:7.0
```

xanderhendriks, 2024-02-28:
Correct. That's the easiest way. 

Are you now able to build in the docker image and is your issue resolved?

krupis, 2024-02-29:
> Correct. That's the easiest way.
> 
> Are you now able to build in the docker image and is your issue resolved?

Yes. There is an example of CI/CD pipeline script to build the project:

```
build_job:
    image: xanderhendriks/stm32cubeide:7.0
    variables:
        GIT_SUBMODULE_STRATEGY: normal
    rules:
        - if: $CI_COMMIT_TAG # Run this job when a tag is created
    stage: build
    script:
        - mkdir binaries;
        - mkdir JOB;
        - stm32cubeide --launcher.suppressErrors -nosplash -application org.eclipse.cdt.managedbuilder.core.headlessbuild -data /tmp/stm-workspace -import . -build my_project/Release

    after_script:
        - cp Release/${CI_PROJECT_NAME}.hex binaries/${CI_PROJECT_NAME}-${CI_COMMIT_TAG}.hex;
        - cd binaries
        - zip ${CI_PROJECT_NAME}-${CI_COMMIT_TAG}_binaries.zip *.hex # zip all .hex files
        - cd ..
        - ls -l
        - cd full_binary
        - ls -l
        - touch JOB_ID.txt
        - echo GE_JOB_ID1=$CI_JOB_ID >> JOB_ID.txt
        - cat JOB_ID.txt
        - ls -l
    artifacts:
        paths:
            - binaries/${CI_PROJECT_NAME}-${CI_COMMIT_TAG}_binaries.zip
        reports:
            dotenv: JOB/JOB_ID.txt
```

The above script is going to build the project and put it as an artifact :)

xanderhendriks, 2024-02-29:
Good to hear you got it working! Thanks for reporting back and closing the issue.