xanderhendriks/docker-stm32cubeide #6: can you planning support new stm32 cube ide 1.15.0 ?

firepooh, 2024-03-28:
hi.

very nice your cubeide git actions.

can you planning support new stm32 cube ide 1.15.0 ?

xanderhendriks, 2024-04-03:
Now supported in version 12 of the [docker](https://github.com/xanderhendriks/docker-stm32cubeide/releases/tag/v12.0) and the [action](https://github.com/xanderhendriks/action-build-stm32cubeide/releases/tag/v12.0)
Please check if it works for you

rena2019, 2024-04-24:
I tried 1.15.0 but it fails while building image:

Step 10/12 : RUN unzip -p /tmp/stm32cubeide-installer.sh.zip > /tmp/stm32cubeide-installer.sh && rm /tmp/stm32cubeide-installer.sh.zip
 ---> Using cache
 ---> aaa190eb22b7
Step 11/12 : RUN chmod +x /tmp/stm32cubeide-installer.sh &&     /tmp/stm32cubeide-installer.sh  &&     rm /tmp/stm32cubeide-installer.sh
 ---> Running in 6c21bdde70a9
Creating directory makeself_dir_Dq4fYV
Verifying archive integrity...  100%   All good.
Uncompressing STM32CubeIDE 1.15.0 installer  100%
Installing STM32CubeIDE into /opt/st/stm32cubeide_1.15.0 ...
Creating directory makeself_dir_cfmAyM
Verifying archive integrity... All good.
Uncompressing STLink udev rules  100%
Package installed.
Creating directory makeself_dir_CF037U
Verifying archive integrity...  100%   All good.
Uncompressing STM STLink-Server installer  100%
stlink-server error while installation started.
Already installed version is newer or equal: line 64:
NOT downgrading. Aborting stlink-server installation.
DISPLAY not set. Cannot display license. Aborting.
Installation failed

my temporary workaround is to exit the stm32cubeide-installer.sh with 0 because I don't care about stlink-server:

RUN chmod +x /tmp/stm32cubeide-installer.sh &&     /tmp/stm32cubeide-installer.sh ; exit 0 &&     rm /tmp/stm32cubeide-installer.sh

xanderhendriks, 2024-04-29:
Interesting. All the versions are specified, so I don't understand why you get a different result than I do. Why not just use the [published docker image](https://hub.docker.com/r/xanderhendriks/stm32cubeide)?
I had it working in my [sample application](https://github.com/xanderhendriks/stm32-sample-application/actions/runs/8737498030).
Please let me know how you go.
@firepooh as original requester, did you manage to test this?

axbusch, 2024-06-04:
works like a charme 👍 

is there a reason why you did not just use the versioning scheme of stm32cube ide? why 12.0 = 1.1x.xx etc

xanderhendriks, 2024-06-13:
Because I wanted to make sure that I could have multiple versions with the same IDE. Happy to change it to another format, as long as that requirement is still met. Could do 1.15.0a? I would only need to know if the tag can indeed contain an extra letter like that. If you can find confirmation of this then I'm happy to implement the change for the next 1.15.1 which is now the current version.

xanderhendriks, 2024-07-10:
As there was no further feedback I'll stick with the current versioning scheme and close this issue.

closed 2024-07-10 by xanderhendriks

axbusch, 2024-07-10:
Hey sorry, totaly forgot.
Was just curious why. For me it's totaly fine as it is 👍

xanderhendriks, 2024-07-10:
Great. Thanks for the feedback.