xanderhendriks/docker-stm32cubeide #8: headless STM32CubeIDE commands to build project don't work - WARNING: No project matched - WARNING: No config matched

yusufso, 2024-07-05:
I am running the two commands you have in the readme. These are the results I get.

Running
`stm32cubeide --launcher.suppressErrors -nosplash -application org.eclipse.cdt.managedbuilder.core.headlessbuild -data /tmp/stm-workspace -import /workspace/PROJECT_NAME`
outputs
`Create.
Opening 'apollo_wifi_hub_stm32'.`

Then running
`headless-build.sh -data /tmp/stm-workspace -build /workspace/apollo_wifi_hub_stm32`
outputs
`WARNING: No Project matched "/workspace/PROJECT_NAME". Skipping...`

I tried several other variations. Another notable one was running (while in the directory /workspace)
`stm32cubeide --launcher.suppressErrors -nosplash -application org.eclipse.cdt.managedbuilder.core.headlessbuild -data /tmp/stm-workspace -build PROJECT_NAME`
outputs
`Project: PROJECT_NAME doesn't appear to be a CDT project. Skipping...
WARNING: No Config matched "PROJECT_NAME". Skipping...`

Any idea what could be going on here? Happy to jump in a videocall to go more in-depth.

Thanks.

yusufso, 2024-07-09:
Found the fix, was something simple. 

When running the first stm32cubeide command, on the import directory you need to add "/STM32CubeIDE" to the end of it. All projects created by STM32CubeIDE (GUI version) will have an folder inside named STM32CubeIDE. So, the project is nested.

When running the second command you only need to write "-build PROJECT_NAME" instead of a directory to the project on the shared volume. The first command already imported PROJECT_NAME to /tmp/stm-workspace and the second command checks for PROJECT_NAME inside this directory.

closed 2024-07-09 by yusufso

xanderhendriks, 2024-07-10:
Great to hear you got it to work. Thanks for letting me know.