xanderhendriks/docker-stm32cubeide #10: Cannot open display when building headless

wouter-heerwegh, 2024-09-24:
Hi,

I'm using this docker image to build an stm32 project headless. I've seen there is an action for it as well, but I'm using a folder I need to include  that is not in the default Src/ or Inc/ directories.

My workflow looks as follows:
```shell
jobs:

  build:
    runs-on: ubuntu-latest
    container: xanderhendriks/stm32cubeide:9.0

    steps:
    - name: Update git
      run: |
        apt-get update
        apt-get install git -y

    - uses: actions/checkout@v3
      with:
        submodules: true
        token: ${{ secrets.TOKEN }}

    - name: Build project
      run: |
        stm32cubeide -nosplash -application org.eclipse.cdt.managedbuilder.core.headlessbuild -import . -build all ${{ inputs.extra_includes }}
```

When using version 9.0 of the docker container, everything works fine, but I've tested all newer versions (up to 13.1), and all of them give the following error:
```
Stm32cubeide: Cannot open display
```
Does anyone know if there has there been a change in the way I should call a headless build?

wouter-heerwegh, 2024-09-24:
I found [this issue](https://community.st.com/t5/stm32cubeide-mcus/stm32cubeide-1-15-1-headless-build-is-not-working/m-p/683242) on stm's forums. Apparently you can't run the command inside the project folder anymore.

I modified my workflow to the following:
```shell

jobs:

  build:
    runs-on: ubuntu-latest
    container: xanderhendriks/stm32cubeide:13.1

    steps:
    - name: Update git
      run: |
        apt-get update
        apt-get install git -y

    - uses: actions/checkout@v3
      with:
        path: ./project
        submodules: true
        token: ${{ secrets.TOKEN }}

    - name: Build project
      run: |
        cd ..
        stm32cubeide -nosplash -application org.eclipse.cdt.managedbuilder.core.headlessbuild -data ./project -importAll . -build all ${{ inputs.extra_includes }}
```
and it seems to work now, so I'm closing this issue

closed 2024-09-24 by wouter-heerwegh

xanderhendriks, 2024-09-25:
I'm glad you got it to work for you. I normally use the action as shown in this repo https://github.com/xanderhendriks/stm32-sample-application

wouter-heerwegh, 2024-09-26:
Hi @xanderhendriks, thanks for the feedback. I indeed saw the action but I would need to do some testing to see if it would build with an extra (non-standard) folder inside the Core folder. I have a modules folder there that contains submodules of reusable code that also needs to be included.

xanderhendriks, 2024-09-26:
It uses the original project file. So anything you configure Stm32CubeIde, which results in a working build, will also work with the action. We have multiple extra source and include directories we add to our projects.