xanderhendriks/docker-stm32cubeide #15: Option to fail on missing project

Love-Pengy, 2024-12-19:
This is more of a question than an issue, but is there currently a way to have it return an error code instead of success code on misconfigured/missing project? At the moment if the project can't be found it gives a warning message: `WARNING: No Project matched "<project name>". Skipping...` but will succeed on that job. I would like it to fail

xanderhendriks, 2024-12-22:
I don't quite understand your question. When using the docker you run your own commands, so you can just add an extra command to check if the expected output is there. Or are you using [the github action](https://github.com/xanderhendriks/action-build-stm32cubeide), which is in another repository.

Love-Pengy, 2024-12-22:
My apologies I am using the action, I will close this and resubmit it in the right repository

closed 2024-12-22 by Love-Pengy