xanderhendriks/docker-stm32cubeide #16: Should probably document that this image doesn't actually run the IDE

hachi, 2025-05-06:
the entry point is just a shell inside the built container, and running the IDE doesn't work.

xanderhendriks, 2025-05-09:
It does run the IDE, it just does it in headless mode. I assume you are trying to run it with the GUI?
This is now supported in [v15.1](https://github.com/xanderhendriks/docker-stm32cubeide/releases/tag/v15.1)
I have instructions on how to run it from the command line if you would like to get the GUI to start right away. It normally goes to the shell as people use this docker image to run CI builds in the cloud.
Please let me know if this works for you.

xanderhendriks, 2025-09-07:
Unfortunately, never received any feedback from the @hachi. I tested it on my machine and it works for me, so closing this issue.